nophead/Mendel90 #58: Frame martial thickness affect wrong dillhole positions

BadKarmaPete, 2015-05-19:
There is a small Issue when creating your own Mende90 with different frame material. The Postion of a few drillholes is affected by the thickness of the choosen material. So when I create my own Mendel90 with MDF12 plates the drillholes for the fixing blocks moving a little bit up or exactly, away from the edge (around 1mm). This makes it difficult to build the frame.

nophead, 2015-05-19:
Not sure what you mean. The sturdy version uses 12mm MDF and all the screw holes come out in the right place.

BadKarmaPete, 2015-05-19:
Ok I've done a mistake. The frame thickness doesn't affect the hole postion. It only seems that the hole is moving, i've meassured the distance between the center of the hole and the edge. The distance is changing when using different screws with different diameters.
For explanation, I was assuming that the drill holes are aligned with their center, but after printing the drillguide from the sturdy version and holding it together with my printed fixing block I see that their are aligned on the outer diameter.

Thank you for your fast response and sorry for my Issue report.

Greetings from Germany

nophead, 2015-05-19:
Still not sure what you mean exactly. If you make the sheets thicker the gantry fixing blocks will move backwards so the holes will be further back. If you make the screws bigger the fixing blocks will get bigger so I think the hole centres probably move back a little as well.

In all situations the drill templates should mark the centre of the hole. In the case of the versions not made by CNC the fixing blocks have slots but the drill centre should still be in the middle of the slot. The slots allow 1mm adjustment either way in case the drilling is not accurate. I used an optical centre punch and a spot drill to get very accurate holes before I started using CNC.

BadKarmaPete, 2015-05-20:
It's my fault, I thought the whole time that the fixing blocks doesn't change their size and only the holes change the diameter by the choosen screw. And after designing the second Mendel90 with MDF plates I failed to see any difference between the fixing blocks of the Dibond and MDF version, so i don't printed the newer ones and thought that the Dibond version should also fit.

nophead, 2015-05-21:
The Dibond fixing blocks have nut traps for M4 nuts but the ones for MDF have slots for wood screws and washers, so they are quite different.

closed 2016-01-05 by nophead