nophead/Mendel90 #59: Mendel90 Kits

Masterrenny, 2015-07-07:
Hello, 

Not sure how else I can contact you, but do you still offer/sell the kits and if so how much would one cost (UK) or would it be cheaper for me to get the parts myself? 

Do you have a template for the frames and everything else? Or a manual I can follow for it as I can't seem to find one. Or a parts list. 

Regards
Thomas

nophead, 2015-07-07:
Yes we sell full kits for the CNC cut dibond version and some parts for the other versions, see http://forums.reprap.org/read.php?94,196585.

If you want to completely self source and don't have access to a CNC router the MDF sturdy version can be made with drill templates found here: https://github.com/nophead/Mendel90/tree/master/sturdy/sheets and the parts list here: https://github.com/nophead/Mendel90/blob/master/sturdy/bom/bom.txt.

Self sourcing will cost less, but not a lot less, because of one off prices and lots of postage from different sources and it will be a lot more work.

nophead, 2016-01-06:
We no longer sell kits.

closed 2016-01-06 by nophead

guerd87, 2016-03-31:
Do you still sell printed parts? Im looking for a complete set of parts including extruder for MDF version

nophead, 2016-04-01:
Yes we still sell the parts listed here: http://forums.reprap.org/read.php?94,567445