nophead/Mendel90 #61: Dibond crashes at Z_max + X_min

PaulCornelissen, 2015-09-27:
The Dibond version crashes the extruder motor into the into the top z-bar clamp.

This has happened to me a couple of times now. When homing the z-axis while at X-min the extruder motor hits the top z-bar clamp. This prevents the limit-switch from being hit so the motors won't stop running. This causes the z-bar clamp to move out of place and messes up the calibration of the z-length. Steps will also be missed and this causes you to lose the levelling calibration. Also if you're not quick to turn the power of the other z-bar clamp will also be hit. I've replaced them both by now.

Here's a photo of this in practice:
[image]

I don't know if this is a known issue all ready, but I've opened up an issue here because today I found out this bug is also in the models:
[image]

Set these three lines almost at the top of ```main.scad```:

    X = 1 * X_travel / 2; // - limit_switch_offset;
    Y = 0 * Y_travel / 2; // - limit_switch_offset;
    Z = 1 * Z_travel;

nophead, 2015-09-27:
It is a known issue (with no simple fix) but at X_MAX, not X_MIN. You should always home X before Z and then it never happens.

nophead, 2015-09-27:
Actually the start.gcode script provided homes Z first but the end.gcode leaves the extruder at the right, so the next start never crashes. The only way it could crash is with manual actions or if it happened to be printing on the left extreme and was interrupted. Basically it never happens in normal use, which is why I didn't fix it. Hundreds of kits were built before anybody noticed.