nophead/Mendel90 #63: sturdy version with dados instead of fixing blocks, offset for gantry and L/R stays?

dangasaur, 2015-12-10:
I'm going to make a version of the sturdy with 3/4" MDF and instead of using the fixing blocks I want to just put a dado in the base for the gantry and stays.  I also want to put dados on either side of the gantry and base for the stays but that dimension seems less critical.

My plan was to simply increase the height of the gantry by the depth of my dado and reference the drill template from the top.  

I'm trying to decipher the scad files to see where I can echo/print out the offset of the gantry from the front or rear of the base as well as the left and right stays.  Can you provide any guidance to get these dimensions?

I was also planning on just reducing the screw count etc in my head but a way to simply exclude all the fixing blocks would also be helpful.

nophead, 2015-12-10:
Hi,
   Had to Google "dado" as it doesn't have the same meaning in the UK.

   gantry_Y is the Y ordinate of the front edge of the gantry. You would need to add base_depth / 2 to get the distance from the front of the base sheet.

left_stay_x and right_stay_x are the X ordinates of the centrelines of the stays.

The fixing_blocks module defined in scad/main.scad places the fixing blocks and their holes and fasteners, etc. If you make it an empty module I think it will fix the bom and drill templates.

dangasaur, 2015-12-12:
Awesome, I think I've correctly located the gantry and the stays now.
Removing the fixing blocks worked as you said.

Thanks for your help.

closed 2016-01-01 by nophead