nophead/Mendel90 #65: e3D wades block, recessed flat spot

dangasaur, 2016-01-16:
After printing the wades block for the e3d hot end I'm seeing that the hole where the hot end is inserted into the block has a flat spot or is "keyed" yet the hot end itself has no flat spot and is perfectly round.  

I feel like I'm missing something but I cannot see how this is supposed to fit together.  What am I doing wrong?

nophead, 2016-01-16:
It is a truncated tear drop shape to allow it to be printed without support, as are all the horizontal holes. The flat is tangential to the circle, so shouldn't prevent a cylindrical hot end going in.

dangasaur, 2016-01-16:
Ah ok, that makes sense.  This must have just been a bad print.

nophead, 2016-01-16:
Yes if your bridges droop you might need to pare some away with knife.

closed 2016-01-16 by nophead

dangasaur, 2016-01-17:
Just in case anyone comes across this at a later date, I concluded that this particular part had a poor print quality from my supplier.  I also discovered a few additional modifications needed for my e3D 1.75 universal hot end.

1. touch up the mounting hole with a 5/8ths forstner bit
2. drill out the filament hole with a #21 drill bit (.159") to fit the ptfe tubing
3. countersink a 9/32nds hole in the mounting hole to provide space for the ptfe coupler

At some point I will learn SCAD and incorporate these modifications for the 2nd printer I am building.

nophead, 2016-01-17:
You don't really want the gearing that Wades provides for 1.75mm filament. A direct drive extruder is better.

lexmark-benmarks, 2016-01-18:
Based on your comment I ended up having a set of 1:1 extruder gears printed.  My plan is to keep moving forward with what I have at this point and then change to a compatible direct drive extruder as soon as possible.

Thanks again for your feedback.