nophead/Mendel90 #71: Project description

hoijui, 2020-07-10:
Hey! :-)
could you maybe add a github repo description (plus tags), and also describe in the README on top what this project is (about)?
would be appreciated!
robin

nophead, 2020-07-10:
Not sure what you mean. Isn't the project description "A RepRap Mendel variant using sheets for the frame instead of rods"?

hoijui, 2020-07-10:
ouhhh... sorry! :D
indeed... I am not used to the new github look yet, sorry!

though it would be nice to have it in the README too.
should I make a PR for that?

nophead, 2020-07-10:
I have added it to the readme. Why the interest? This is an old project and I don't actively maintain it now.

hoijui, 2020-07-10:
I found to it from the open-manufacturing google group, where it was linked as someones motivation for building a library of standard parts for OpenSCAD and FreeCAD, because it is so well made. :-)

I am from OSE Germany, and we are working on.... stuff to improve the state of open hardware.

nophead, 2020-07-10:
If you want a library parts, see my other repository. https://github.com/nophead/NopSCADlib.

I developed it from the vitamins in Mendel90 but greatly improved them and I actively maintain it. If forms the basis of everything I 3D print and I am using it for my next 3D printer design.

hoijui, 2020-07-11:
oh WOW! :O
thank you!

I myself am a programmer, and thus OpenSCAD is more natural to me. Most of our tinkerers are mechanical engineers though, and they are all FreeCAD oriented. Do you have some way, or an idea, how to use your library there too?

nophead, 2020-07-11:
I think FreeCAD can import a OpenSCAD csg file, which is like the original source code except all the expressions have been replaced by constants and all the modules have been expanded to csg ops on primitives. I don't know if it supports all of OpenSCAD and I think you lose all the parametric aspects.

I think working with conventional CAD like FreeCAD you would import STEP files of components that a lot of manufacturers and other websites publish. They are more detailed and more accurate. I prefer to use OpenSCAD because I am a retired programmer, not a mechanical engineer. I also like the fact the representation of objects is human readable text, so it can be source controlled, etc, and I like the fact everything is 100% parametric by default.